prusa3d/PrusaSlicer #8843: Short extrusion like stuttering (again)

MagicAlien, 2022-09-10:
### Description of the bug

With Arachne engine, I have (again #8511 #8534 ) a lot of short extrusions. No problem with classic.

### Project file & How to reproduce

[contenitor.zip](https://github.com/prusa3d/PrusaSlicer/files/9539952/contenitor.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 11

### Printer model

Anycubic Kossel Linear Plus with marlin 2.0

MagicAlien, 2022-09-10:
[gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/9539959/gcode.zip)

wabbitguy, 2022-09-14:
I have the same problem with Arachne engine as well. Even a flat plate with small mounting holes will stutter on the 3mm holes.
[wide_wheel.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/9569634/wide_wheel.stl.zip)

closed 2022-09-20 by DragonFinlay