prusa3d/PrusaSlicer #8844: Crash slicing with 0.6mm nozzle, MMU

Noggin01, 2022-09-10:
### Description of the bug

PrusaSlicer properly slices a model for printing with a 0.4mm nozzle. When I select a printer profile for a 0.6mm nozzle, PrusaSlicer crashes.

### Project file & How to reproduce

[900-10832-00 - Lid SN0019.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9540416/900-10832-00.-.Lid.SN0019.3mf.zip)
[900-10832-00 - Lid SN0019 6n.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9540417/900-10832-00.-.Lid.SN0019.6n.3mf.zip)

To reproduce the bug:

1. unzip the two project files
2. Open the file "900-10832-00 - Lid SN0019.3mf" in PrusaSlicer
3. Slice the file
4. Open the file "900-10832-00 - Lid SN0019 6n.3mf" in PrusaSlicer
6. Slice the file
7. At this point, PrusaSlicer crashes on my system
8. Open the file "900-10832-00 - Lid SN0019.3mf" in PrusaSlicer
9. Select a 0.6mm nozzle profile
10. Transfer settings
11. Slice the file
12. At this point, PrusaSlicer crashes on my system

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+linux-x64-GTK2-202209060714

### Operating system

Linux Mint 20.2 Cinnamon; Kernel 5.4.0-124-generic

### Printer model

Prusa i3 MK3S+ w/ MMU2S

FidelCapo, 2022-09-12:
I think it's the same issue as https://github.com/prusa3d/PrusaSlicer/issues/8837

rtyr, 2022-09-14:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8837 (and it is already fixed in master).

closed 2022-09-14 by rtyr