prusa3d/PrusaSlicer #8845: Feature request: SLA supports for FDM prints opt-in feature

mcurrao, 2022-09-11:
**Is your feature request related to a problem? Please describe.**

Proper FDM tree supports are unavailable still. 
Even once released, they may not be as useful for really small miniatures as SLA's tree supports once released. If they behave like Cura's supports, FDM tree supports may be still to chunky for miniatures.

**NOTE:** This is not a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1514, but rather an alternative solution/workaround request in the meanwhile.

**Describe the solution you'd like**

Allow enabling SLA supports for FDM prints. Not FDM-optimized tree supports, but rather SLA supports as currently available.

**Describe how it would work**

- Add a boolean property in the `Configuration -> Preferences -> General` menu with a caption similar to "Enable SLA supports for FDM printers". This property would be unchecked by default.
- Add a message below said menu stating that said property is not recommended, and is an "advanced feature".
- If the property is checked, the "SLA Support Points (L)" menu is displayed regardless of the printer type.
- Pads are not used, and the standard "brim" of FDM supports can be used as a replacement if required.

**Describe alternatives you've considered**

- Having proper FDM tree supports (unavailable; requested since 2018; issue has been blocked and no release date has been provided).
- Keep using the workarround of simulating an SLA print, exporting, importing again. Cumbersome; not ideal.

**Additional context**

#1514 has been open since 2018, and is currently blocked. That means that even reacting/upvoting is unavailable.
Considering FDM tree support seems to involve extra difficulty, **could we consider adding a disabled-by-default option in the properties menu to display the SLA tree support even on FDM?**

Note that tree supports has been asked for way too long and way too many times:

- https://github.com/prusa3d/PrusaSlicer/issues/8041
- https://github.com/prusa3d/PrusaSlicer/issues/7927
- https://github.com/prusa3d/PrusaSlicer/issues/6880
- https://github.com/prusa3d/PrusaSlicer/issues/5329
- https://github.com/prusa3d/PrusaSlicer/issues/4376
- https://github.com/prusa3d/PrusaSlicer/issues/1681
- https://github.com/prusa3d/PrusaSlicer/issues/1514 // The blocked one 🚫 

I understand FDM tree supports are different for large scale prints, as they need to be hollow, wider, etc. I also understand this involves extra development, in order to build a new support-processing engine for FDM printing.
But for miniatures, SLA-style-tree support can be sometimes even more practical for FDM than FDM's own tree support. 

Users are already using SLA supports by adding a fake SLA printer, exporting the supported structure, and reimporting the supported model. A way of doing this without having to add a fake SLA printer would be greatly appreciated.

gudvinr, 2022-09-13:
SLA supports have different purposes. While FDM supports just need to provide surface for plastic to extrude, SLA supports are used to overcome pulling and squishing force caused by liquid because parts printed upside down as opposed to FDM prints.

Physics of material deposition is completely different so this can't be "an alternative" because these supports simply won't work as expected.

If you read #1514 you will notice that they explicitly said:
> As I mentioned recently, we are actively working on this. The polishing will still take a few weeks. Thank you for your patience.

It was closed in first place because comments like yours.

lukasmatena, 2022-09-14:
We do not plan to enable SLA supports in FDM mode. Definitely not with current status of FDM tree supports development. The reason is the same as described in #8318. We want to finish the real solution, not waste time on workarounds. The steps that you describe are not as trivial as they sound, for example the SLA Support Points menu would not work with multipart objects, the SLA and FDM backends are completely separated, so it cannot be just "enabled", etc.

closed 2022-09-14 by lukasmatena