prusa3d/PrusaSlicer #8846: Mysterious Perimeter and Overhangs appear with Arachne

MartinMajewski, 2022-09-11:
### Description of the bug

Printing features that are in the range of two perimeters printed only (no infill needed, the wall-thickness is only two perimeters), mysterious glitches appear. These glitches show as perimeters and overhanging perimeters where none should be.

I have attached pictures of the object and the sliced result.

[image]
[image]
[image]

### Project file & How to reproduce

[IssuesWithMysteriousPerimeters.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9542321/IssuesWithMysteriousPerimeters.3mf.zip)

Please find attached the project file.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 - Release Version

### Operating system

macOS 12.5.1

### Printer model

Prusa MK3S

MartinMajewski, 2022-09-11:
Btw.: this glitch appears as shown in the screenshots with the 0.6mm nozzle profile. Changing the nozzle profile to 0.4mm just changes the expression of the glitches but does not remove them.
[image]

MartinMajewski, 2022-09-12:
Here is the wrongly produced gcode (on a newer version of the model):

[image]

hejllukas, 2022-09-13:
Thank you for reporting this issue. Your issue is caused by an issue in the Boost Voronoi diagram generator library that Arachne uses. The boost Voronoi diagram generator is producing a non-planar Voronoi diagram that isn't correct.
It should happen only in very rare cases, but we are investigating this issue.

For now, you can workaround this issue by any of those two workarounds:
- Slightly rotate (1° is enough) the model around the Z axis.
- Change Perimeter generator settings from Arachne to Classic (The model will be sliced in the same way as in 2.4.2).

The G-code line that you mentioned is correct. G-code specification allows instead of -0.05 to produce just -.05, and printers process it correctly. We started emitting those G-code lines in PrusaSlicer 2.4.0 release.

benkrejci, 2022-09-15:
Here's an extreme example which I assume is caused by the same thing:
[image]

If you want to test with it, here's the file:
[120mmFanDesiccant.zip](https://github.com/prusa3d/PrusaSlicer/files/9571148/120mmFanDesiccant.zip)

Default prusament petg profile
Default prusa mini+ 0.6mm nozzle profile
Default settings with arachne

PrusaSlicer
Version:   2.5.0+win64
Build:     PrusaSlicer-2.5.0+win64-202209060714

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 34,301MB
OpenGL installation
GL version:   4.6.0 NVIDIA 516.59
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3070 Ti/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
[... 333 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

garethky, 2022-10-30:
+1, spotted this in one of my parts with x/y planar "thin" walls. Its producing infill in these areas and then producing strange perimeters that exit the model on top of the infill. Subscribed.
[image]

[Sorting Box Support Rails.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9897046/Sorting.Box.Support.Rails.3mf.zip)

kubispe1, 2023-01-17:
These problems related to the Voronoi diagram generator are fixed in the next release 2.6.0. Here, I attach pictures and close the GH issue as resolved.

[image]

[image]

[image]

closed 2023-01-17 by kubispe1