prusa3d/PrusaSlicer #8850: Incorrect display STL file name if not western coding page characters used

BoevoyNarod, 2022-09-11:
### Description of the bug

Incorrect display STL file name in the "view model filename" option if not western coding page characters used (Cyrillic) and there are spaces in the file name.
[psconfig.ZIP](https://github.com/prusa3d/PrusaSlicer/files/9542910/psconfig.ZIP)
[image]
[image]
[image]

### Project file & How to reproduce

Activate view-show model name and choose Cyrillic file name with spaces

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+win64-202209060714

### Operating system

Windows 10

### Printer model

It was born as Ender 3 pro

FidelCapo, 2022-09-12:
What language do you have set in PrusaSlicer? Ukrainian?

BoevoyNarod, 2022-09-12:
Russian Отправлено из мобильной Яндекс.Почты: http://m.ya.ru/ymail

FidelCapo, 2022-10-20:
This issue should be fixed and the fix will be in version 2.6.0. 
Closing.

closed 2022-10-20 by FidelCapo