prusa3d/PrusaSlicer #8852: New feature : wave dense support or sawtooth pattern

AnonymousXeoiPi, 2022-09-11:
Is your feature request related to a problem? Please describe.
Not easy to remove support from parts

Describe the solution you'd like
Playing with dense support layer, we could add z variation (parameter from 0.05 to 0.2) or pattern that make peak before part layer. This will reduce surface contact between support and part to be able to remove supports with ease.

Describe how it would work
By applying a sinusoïdal fonction in z direction we can reduce surface contact between dense support and part. Each line could be glide by a half distance of peak to peak to get a more or less dense peak surface.

3DCoded, 2022-09-11:
What do you mean not easy to remove support from parts? What printer are you using? On my Prusa MK3S+, the supports come off just fine.

AnonymousXeoiPi, 2022-09-12:
At present, the support is a platform with a large contact area，Even increasing the spacing is a line. If it is a serrated support then the point connection contact area is smaller and easier to remove. At the same time, the contact Z distance can be smaller to obtain a better surface. Like sawtooth pattern from Superslicer. 
[image]
 The printer model used is irrelevant.

closed 2026-09-29 by prusaslicer-bot[bot]