prusa3d/PrusaSlicer #8853: Export Model with modifiers as STL

Meta-Cake, 2022-09-12:
Version 2.5.0
Windows 10 Pro

**Is your feature request related to a problem? Please describe.**
Whenever I have a model with a stamp; such as the one on the bottom of the Benchy, and I want to remove it inside of Prusa and replace it with a different one using imported STLs as a negative volume instead of launching a modeling software; I am unable to save the model as an STL; if I am adding something to the model such as adding a part I can export it fine, but the negative volume is not exported with the model and I am forced to edit the model 

**Describe the solution you'd like**
The ability to "merge" the negative volume to the model/models (including added parts nested to the main part that can't be merged directly to the main part for some reason) the same way boolean works in most modeling software 

**Describe how it would work**
selecting the negative volume modifier and any pieces desired to be affected by it (non-selected parts that are in the space the negative volume occupies should be excluded) then right-clicking and selecting merge just like how you would merge two objects normally

rmeden, 2022-11-23:
Is this a feature or a bug?  The fact that modifiers are not included in the STL sounds like a bug to me.  I know the STL export didn't work as I expected.

In my case I was designing a compound washer (for a toilet bolt).  It's such a simple design I used shapes to do it.  The hole in the center doesn't make it to the exported STL.  I'll attach the model.
[toilet_washer.zip](https://github.com/prusa3d/PrusaSlicer/files/10072495/toilet_washer.zip)

vintagepc, 2022-12-02:
I too noticed this when using negative volumes to modify an STL for which I could not edit the sources - the regions I'd added them were just solid in the export.

VH-fixer, 2023-01-09:
I know the slicer is not a CAD system, but CAD systems still do not handle .stl files well.  The vast majority of files we share are .stl or similar low data files.   Mods are part of our everyday life. 

We can merger  two or more .stl files together and then export the new .stl file.  It is a great feature.
We can generate the correct G code when there is a negative volume.  This is great.

We cannot merge a "negative" volume and export the new .stl.   This is a major problem.   We can only do half the job of creating the many mods we need to do.   This is a MAJOR functional gap in the slicer.   This is like driving a car and only being able to go straight or turn left.  No ability to turn right. 

When negative volumes are involved, I have to keep a slicer project file.    It is too easy to change the wrong thing with a complex assembly of .stl files with negative volumes.  CTRL Z or just exiting without saving is done often.    A complete .stl file definition is much better for sharing and storage.  A project file may become out of date or may not be a "mistake free" option when sharing with others.

jwkovell, 2023-01-21:
Yes please! With the announcement that PrusaSlicer will soon support customizing models with text, this request gains even more relevance. 

The more models can be altered within the slicer, the more often situations where "baking" those modifications into a new stl mesh arise.

Saving the project as a 3mf is good, but not sufficient in many cases.

lukasmatena, 2023-03-01:
@Daelilian 
> why can't there be a dropdown option
> such a simple solution

The problem to solve is not a missing dropdown. The problem is that the "subtraction of meshes" is not as trivial as it sounds and it is simply not implemented yet. Shortly discussed in #7594 (which this one is a duplicate of).

Daelilian, 2023-03-01:
It would be the same as the difference boolean on blender.  blender is open source as well. I know it may be c complicated when it comes to the code languages. but some of blender uses c++ so maybe some of that code could be implemented?

kubispe1, 2023-03-15:
We are looking into this issue. 
SPE-1540

lukasmatena, 2023-04-02:
Subtraction of negative meshes during export was implemented in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena