prusa3d/PrusaSlicer #8854: [PS 2.6.0a0] Rotating Mirrored Part Flips Back To Normal Orientation

foreachthing, 2022-09-12:
### Description of the bug

When rotating a mirrored part, it flips back to not-mirrored state.

### Project file & How to reproduce

1. add shape gallery; pick the bunny
2. add another bunny through the gallery
  2.1. maybe scale it down, too
1. mirror _one_ bunny around **X**
1. ROTATE the just mirrored bunny (i.e. use the textbox on the right to enter `1` into the Z-box)
1. compare bunnies

Before rotation (after mirroring):
[image]

After rotating the green bunny 1° around Z:
[image]
**Bunnies are now the same at different rotation!**

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha0-156-gc3bdf2fc2

### Operating system

Windows 10 x64

### Printer model

n/a

kubispe1, 2022-09-14:
I can see you use 2.6.0-alpha0-156-gc3bdf2fc2. Does this problem also appears in our standard release 2.5.0? There it looks quite ok. Thanks

foreachthing, 2022-09-14:
> Does this problem also appears in our standard release 2.5.0?

Nope! 2.5.0+Win64 behaves as expected.

kubispe1, 2022-09-14:
We detected exactly this kind of bug in previous version, but it should be fixed now and also in upcoming releases. Maybe in some branch it could persist.

foreachthing, 2022-09-14:
I just rebuild the whole shebang from `git clone https://github.com/prusa3d/PrusaSlicer.git --recursive`.
But still no change....
So, if you think you want to close, go ahead. I'll check back as soon as 2.6 is out.

kubispe1, 2023-02-02:
Hi @foreachthing, can you please check this issue with 2.6.0? a2, if it is fixed. Thank you so much

closed 2023-02-03 by foreachthing