prusa3d/PrusaSlicer #8855: Concentric infill on the bottom of supported object

HonzaPz, 2022-09-12:
**Is your feature request related to a problem? Please describe.**
I found that concentric pattern on the bottom of a supported object provides better print quality in certain cases. However, PS doesn't provide an easy solution to change the contact layer infill pattern. (Or if there is the setting, I couldn't find it)

**Describe the solution you'd like**
A toggle in the infill or support settings, which would allow to change infill pattern, similar to current interface pattern setting.

**Describe alternatives you've considered**
Similar effect can be achieved with use of modifiers. However, it is only usable on simple objects.

**Additional context**
Comparison of same object with same (not optimal) support settings, but different pattern:
[image]

kubispe1, 2022-09-14:
Thank for your effort. I think that needs more investigation before implementation. In case you will have more results collect it here. :D

mactul, 2022-09-15:
I think I have a partial solution for you.
One of the reasons that make your supports better is that layer adhesion is'nt great between a concentric layer and a rectilinear layer.

Instead of changing the bottom layer of your print, you can change the top layer of your supports. PrusaSlicer allows that.

I don't have my computer with me, so I can't send you my exact supports settings now.

But I think I use 4 concentric interface layers on the top of the supports, with 90% density.
I let a gap between the top layer of the support and the bottom layer of the print. I don't remember exactly, but maybe 0.15mm for a 0.2mm layer height print.

I also recommend that you reduce the print speed of the interface layers. This will help to have a more precise support.

HonzaPz, 2022-09-15:
Thanks for the responses.
It's not that I have issues with supports, but it surprised me how much of improvement can be reached just by switching to concentric infill, while still being easy to remove. Mechanical parts often have large flat areas that need supports, so I thought It'd be worth investigating. I'll try to do few more tests.

HonzaPz, 2023-02-02:
The feature is available in 2.6 Alpha 2 as "Extra perimeters on overhangs" function.

closed 2023-02-02 by HonzaPz