prusa3d/PrusaSlicer #8857: Problem with thin vertical lines.

WCubitt, 2022-09-12:
### Description of the bug

When sliced in 2.5.0 thin vertical lines become jagged, when sliced in 2.4.2 this is not a problem.
[image]
[image]

### Project file & How to reproduce

[MMU peg.zip](https://github.com/prusa3d/PrusaSlicer/files/9547339/MMU.peg.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 64bit

### Printer model

Mk3s

neophyl, 2022-09-12:
As you havent included the requested project file and only the gcode (which is not very useful for debugging these types of issues) we can only guess.  
However my best guess would be to change the slicing engine back to classic instead of Arachne.

WCubitt, 2022-09-12:
Sorry saved copy of both files and attached wrong one.
Slices correctly in Classic mode.
[MMU2S Tension Screw Retainer & Thumbscrew up.zip](https://github.com/prusa3d/PrusaSlicer/files/9548756/MMU2S.Tension.Screw.Retainer.Thumbscrew.up.zip)

malibata, 2022-09-17:
As  mentioned (erroneously) in other post https://github.com/prusa3d/PrusaSlicer/issues/8824 ...
This issue seems to be caused by the default setting of "Advanced/Arachne.../Perimeters transition threshold angle" that is set to 10 degrees, increasing the value to 15 degrees fixes the issue for me.
Maybe the default should be changed to 15 because, at  least for me, 10 is causing similar issues quite often ...

closed 2026-09-29 by prusaslicer-bot[bot]