prusa3d/PrusaSlicer #8859: Prusa Slicer 2.5.0 - Significantly reduced network transfer speed

oriddlero, 2022-09-12:
### Description of the bug

immediately after upgrading to prusa slicer 2.5.0 network transfer speeds when transferring G code to my multiple MKS3+ printers was reduced to a crawl.  For example: a 20 MB G code file used to take about three or four seconds in the previous version, now that same file can take close to a minute, exponentially slower.

Nothing has changed except for the slicer upgrade on my end.  I'm Running Octoprint, on RPI Zero2w's on 2 Prusa MKS3+ printers, both exhibit the same symptoms after upgrading.

### Project file & How to reproduce

1. do a side-by-side comparison of network transfer speeds on the previous version of prusa slicer versus 2.5.0 to attempt to replicate the issue.  

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

W10 Latest

### Printer model

MKS3+

DragonFinlay, 2022-09-20:
I can't reproduce this issue on my end, upload speeds to Octoprint are still the same as before.
Did you try re-installing the previous version of PrusaSlicer you were using and checking if you get the fast upload speed again? Just so we can be certain it really has to do with 2.5.0.

kocikdav, 2022-09-20:
@oriddlero Thank you for reporting this, unfortunately we have never experienced such slowdown in any situation and network we were testing. What version of PS you were using before? Would you try that version again? Couldn't there be any other reason for the slowdown on your network?

closed 2026-09-29 by prusaslicer-bot[bot]