prusa3d/PrusaSlicer #8861: Variable Layering Height and nozzle diameter

laptop-laptop, 2022-09-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Hi, usually I print at the default nozzle diameter of 0.4 mm on my Prusa i3MK3S+. I changed the nozzle diameter and extrusion width to correspond to 0.6mm diameter nozzle according to:

https://help.prusa3d.com/article/creating-profiles-for-different-nozzles_127540

However, after slicing, the estimated print time is only one minute less than that to be printed by a 0.4mm nozzle. Shouldn't larger diameter nozzle take less time to print?

Before slicing, I used the Variable Layering Height function. Supposing that I print using a 0.4mm nozzle with layer height 1.5mm. I can understand that a larger diameter nozzle would take less time to complete the print job. How does it work when Variable Layering Height is used? Could using this function makes the time to be saved by larger diameter nozzle negligible because this function causes the printing of some layers thinner (in some cases much thinner) than 0.4mm so printing with a nozzle with diameter larger than 0.4mm takes the same amount as printing at 0.4mm?  How does it work?

gudvinr, 2022-09-15:
> Shouldn't larger diameter nozzle take less time to print?

No, bigger nozzle doesn't work on magic. You don't get less time simply because of the nozzle size. It just gives you opportunity to physically push more filament.

You get less time because instead of e.g. 3x0.4mm walls you can print 2x0.6mm walls which cuts time because you travel less. 
Also you there's less back pressure which allows you to increase printing speed (if your heater can keep up) and layer thickness more.  
Although you still need to consider max volumetric flow that your hotend+extruder combo can provide because with bigger nozzle it is easier to hit limits.

Also 1.5mm is kinda too big of a height for 0.4mm nozzle but since your example is hypothetical I don't think it matters. But your assumptions are wrong, simply installing new nozzle won't make you print faster.

gudvinr, 2022-09-15:
> What is "back pressure"?

There's wikipedia article that is named exactly like that.

It's the force directed in the opposite direction of desired filament flow (which is down through the nozzle). Since output is restricted, there is limit on how much you can squish semi-liquid plastic. More restricted outlet is more force there is. And vice versa.  
Term is not correct since pressure can't have direction but it doesn't matter. Main point is that bigger nozzle only gives you _possibility_ of producing prints at less time and does not make printer print faster just because you have bigger nozzle.

laptop-laptop, 2022-09-17:
Thank you for your clarifications.

DragonFinlay, 2022-09-20:
For any more help regarding this, please contact our tech support at info@prusa3d.com, or on the LiveChat in the bottom right corner of the [e-shop](https://www.prusa3d.com/).

closed 2022-09-20 by DragonFinlay