prusa3d/PrusaSlicer #8862: Flow::spacing() produced negative spacing. Did you set some extrusion width too small?

404hult, 2022-09-12:
### Description of the bug

I can not slice models with support structures. I have tried paint-on supports and also autogenerated on two different models. I get an error message "Flow::spacing() produced negative spacing. Did you set some extrusion width too small?" (I have an old firmware on my Prusa MK3 printer, 3.8.0)

### Project file & How to reproduce

[image]

[knivhallare_victor_liten.zip](https://github.com/prusa3d/PrusaSlicer/files/9550893/knivhallare_victor_liten.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

windows 10 Pro

### Printer model

Prusa MK3S_MMU2

natthapolvanasrivilai, 2022-09-13:
Enable the "Thick bridge" for the problematic part seem to solve the issue.

404hult, 2022-09-13:
Unfortunately it did not work. Same error. I tried both enabled and unabled thick bridges.

404hult, 2022-09-13:
*disbled

DragonFinlay, 2022-09-20:
I can reproduce this, and it seems to be related to your printer profile. It has a lot of questionable changes, some of which can definitely be the cause of this error.
The cleanest solution would be to go back to using the system profile and if needed, redo any changes that are necessary for you.
Feel free to let me know if that works for you.

404hult, 2022-09-20:
Thank you for the feedback.

I will explore resetting the Printer profile.

Best Regards
Stefan Hult

yapalexei, 2024-11-01:
fyi, I had a similar issue that occurred for me when I enabled multi-material printing. And what fixed it for me was setting the outer wall line width parameter to a none-zero value. My flavor of slicer, based on prusa, by default, had set all the line width values (except the default width) to zero. I suppose when a parameter value is 0 it will "default" to what the "default" value is. And this had worked as such until multi-material printing was enabled.

However, after looking at the code that emits this exception/error, I think the message could, at least, be a bit more descriptive to help the user recover from their mistake by setting the "outer_wall_line_width" to a none-zero value.

anyjomo, 2025-02-12:
From my understanding, the error comes from the Flow class:

Flow/Spacing Constraint 
```
FlowErrorNegativeSpacing::FlowErrorNegativeSpacing() : 
	FlowError("Flow::spacing() produced negative spacing. Did you set some extrusion width too small?") {}
...

float Flow::rounded_rectangle_extrusion_spacing(float width, float height)
{
    auto out = width - height * float(1. - 0.25 * PI);
    if (out <= 0.f)
        throw FlowErrorNegativeSpacing();
    return out;
}
```
the error condition is:

` out = width - height * float(1. - 0.25 * PI) <= 0

or more or less

out = width - height * 0.2146 <= 0

so for your conditions to hold,

width - height * 0.2146 > 0
`

the **height** is controlled in the **layer height** and **first layer height** configurations, under _Print Settings_
the **width** is 1.125f * nozzle_diameter if you choose the option _Automatic extrusion widths calculation_ option under Print settings/Advanced for the Solid Infill, the external perimeter, the perimeter, and infill. And is equal to the nozzle_diameter for the supports and top solid infill layers. 

meaning that in the worst case scenario you can consider:
 ` out = nozzle_diameter - height * float(1. - 0.25 * PI);
`

If you don't use the automatic option and instead set the widths manually, they are derived from the nozzle diameter, but you still have to hold the error condition expression.

 ---------------------

Also, you have to respect a seccond constrain, lets call it the Layer Hight constrain which yields the error 
"Layer height can't be greater than the nozzle diameter" which specifies that:

Default Maximum Layer Height (0.75 * nozzle_diameter):
if you set the maximum layer height to 0 in the Printers/Extruder/Layer height limits

Hope this helped.

----------------------
references:
https://help.prusa3d.com/article/layers-and-perimeters_1748
https://manual.slic3r.org/advanced/flow-math