prusa3d/PrusaSlicer #8863: Brim width around support structures does not reflect manually set brim width for object

Bushmills, 2022-09-12:
### Description of the bug

For a somewhat problematic ABS part I increased width of brim around the object. The object also needed support structures. After slicing, I notices that the width of brim around support structures remained unchanged, also there seems to exist no manual setting or override for those - though I may be mistaken on this. 
It is my opinion that set brim width should apply to any structure, be it support or parts of object layout.

### Project file & How to reproduce

set brim, increase brim width
enable generating support
slice

notice that while brim around object increases in width, brim around support doesn't

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Linux (Debian bookworm)

### Printer model

Anycubic Vyper

neophyl, 2022-09-13:
It’s under the support settings, called something like expansion, set to 3mm by default.

rtyr, 2022-09-14:
Brim is generated around the object.

Go to support material setting and increase first layer expansion value as neophyl suggested.

closed 2022-09-14 by rtyr

Pigjes, 2022-10-11:
I have a strange brim - raft issue too in 2.5.0. I print miniatures mostly, and sometimes, it does not surround or go under the entire object or even supports, but in a weird position on the left only. Not on a corner, but in the middle of one side. Like a small blob. This does not suffice. I slice it in 2.4.2 and then it's correct.

DragonFinlay, 2022-10-12:
@Pigjes Are you sure the model is nice and flat on the bottom? Can you attach the 3MF so we can have a look?

Pigjes, 2022-10-12:
@DragonFinlay Yes, perfectly flat, several files, all miniature STL's. When I throw them in 2.4.2, no problem whatsoever printing it with full brims instead of these meager blobs 2.5 throws at them. Not uploading my STL's though, as I create miniatures myself and they are not for public use. I have been printing them for over a year now, this is a first.

DragonFinlay, 2022-10-13:
I understand. Perhaps you can use the "simplify model" function in PrusaSlicer to get rid of all the details (so nobody can use the model for their own purposes) and then have it attached here? Given that the brim issue still exists after simplifying of course.