prusa3d/PrusaSlicer #8864: 2.5.0 Windows 11 frozen pixels and crashing

FilamentFrenzy, 2022-09-12:
### Description of the bug

Hi, since updating to 2.5.0 im experiencing a glitching screen on windows 11 and consistent crashes, ive attached a link below to a video showing what im experiencing, thanks, tom

### Project file & How to reproduce

Video Link:
https://1drv.ms/v/s!AkVkL63GMSEogc1zdVavcA-lg_lSFw?e=OJrxgx

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 11

### Printer model

all modles

kubispe1, 2022-09-21:
Hi, can you provide us some technical information of your machine. Especially graphics card info.
For case, you can also attach your .3mf.... But technical info is more important. Thanks a lot.

FilamentFrenzy, 2022-09-21:
Hey,Please see machine specifications belowDevice name	Omen Processor	AMD Ryzen 7 5800H with Radeon Graphics 3.20 GHzInstalled RAM	16.0 GB (15.3 GB usable)Device ID	CE602234-8100-4582-A632-25CF025F2701Product ID	00342-20770-78853-AAOEMSystem type	64-bit operating system, x64-based processorPen and touch	No pen or touch input is available for this displayEdition	Windows 11 Home Version	21H2Installed on	‎04/‎05/‎2022OS build	22000.978Experience	Windows Feature Experience Pack 1000.22000.978.0Hope that's the info you need if you require anything else let me know, ive also attached 3MF but this happens with any file and any profileThanksTom

closed 2026-09-29 by prusaslicer-bot[bot]