prusa3d/PrusaSlicer #8865: [Typo] Typo in the description

natthapolvanasrivilai, 2022-09-13:
### Description of the bug

There is a little typo in the extrusion width description.
[image]

Not a major issue, but leave it here as I found it.

### Project file & How to reproduce

Go to Extrusion Width settings, and hover the mouse over **Infill width** setting.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 11

### Printer model

MK3/MK3S

neophyl, 2022-09-13:
Hate to tell you but Extrudate is a proper word.

Noun
extrudate (plural [extrudates](https://en.wiktionary.org/wiki/extrudates#English))
Material that has been [extruded](https://en.wiktionary.org/wiki/extrude) through a die

rtyr, 2022-09-13:
Answered. Closing.

closed 2022-09-13 by rtyr