prusa3d/PrusaSlicer #8866: Infill to wall distance

Perrow, 2022-09-13:
Much like you can set a distance from your model to supports I would like a similar setting for infill.

I'm currently printing a Raspberry Pi-enclosure (https://www.thingiverse.com/thing:1572173). It has many long openings in the lid and bottom but also larger surfaces. With three top and bottom layers and three perimeters there's both areas that needs the support that infill gives but many where infill will make no difference at all.

I think it would work if the infill algorithm looks at the bridging of the first top layer and only infills beneath sections where there will be more than X mm of bridging. There would ideally be some logic to make sure there won't be thin, high, pillars of support

neophyl, 2022-09-13:
Have you tried the Only infill where needed option ?  
Print Settings>Infill>Reducing Print Time>Only infill where needed

th33xitus, 2022-09-17:
This would be a fairly simple usecase for modifiers. Place a modifier over the area where you don't want infill to be and set the infill to 0%. There is really no need for a dedicated slicer setting imho.

DragonFinlay, 2022-09-17:
I agree, this is one of the reasons why we have modifiers. I understand that sometimes it can be tedious if you have to place a lot of modifiers, but for now, I don't think a separate setting would be necessary. Possibly in the future.

closed 2022-09-17 by DragonFinlay