prusa3d/PrusaSlicer #8867: Aligned seam is not so aligned

ochm, 2022-09-13:
### Description of the bug

Aligned seam puts seam in one place, but in one layer is seam at other place and o do not see why, IMHO bug.
In attached project [test.zip](https://github.com/prusa3d/PrusaSlicer/files/9558134/test.zip) layer 41, 
shown in attached screenshot [image]

### Project file & How to reproduce

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/9558134/test.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 Win64

### Operating system

Windows 10

### Printer model

Prusa MK3 S

DragonFinlay, 2022-09-21:
It seems to be related to the model geometry/topology. When I NetFabb-fix or simplify the model the problematic seam moves, but it never aligns with the rest of the seams.

dkbay, 2025-04-09:
I have a similar issue.
Got no clue why the seams suddenly get randomly placed around the model.
Makes it end up looking quite bad.
I've worked around it by setting the seam position to rear for now.
But ideally aligned seams should be aligned.

[image]

dkbay, 2025-04-10:
Hey @SachCZ would prefer not to share the model here publicly.
Using v2.9.1 of the slicer.

SachCZ, 2025-04-10:
Hello, thank you, we understand that. If anyone is able to replicate the issue, it would be of great help to us.

closed 2026-09-29 by prusaslicer-bot[bot]