prusa3d/PrusaSlicer #8868: Repair button remains after model is reloaded with no errors

markuskruse, 2022-09-13:
### Description of the bug

Background
A model with errors are loaded. The repair button (triangle with exclamation mark) is shown next to the file name.

**Action**
The model is repaired outside Prusa slicer then reloaded.

**Expected outcome**
Repair button is no longer visible (or clickable).

**Actual outcome**
Repair button is still visible and clickable.

### Project file & How to reproduce

[config.zip](https://github.com/prusa3d/PrusaSlicer/files/9558257/config.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11 64bit

### Printer model

Prusa Mk3S

markuskruse, 2022-09-13:
This behaviour has been in the application since before 2.5.0. From what I have been able to see, it does not matter for what printer I slice (I don't even have to slice).

rtyr, 2022-09-14:
Please provide your 3MF project file.

markuskruse, 2022-09-14:
[image]
[cube-broken-and-fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/9565227/cube-broken-and-fixed.zip)

Instructions: 

1. Load the cube.stl
2. Check that it is broken.
3. Remove it from the filesystem. Keep it open in the slicer.
4. Rename the cube-fixed.stl to cube.stl.
5. Right click the model in the slicer and select "reload from disk"
6. Check that the new model is free from errors and that the repair icon is still visible.

markuskruse, 2022-09-14:
[cube-broken-3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9565336/cube-broken-3mf.zip)

DragonFinlay, 2022-09-21:
The 3MF you provided only includes the "cube.3mf", so I can't follow the steps you listed. In the given 3MF I can successfully repair the cube and the triangle is gone.

markuskruse, 2022-09-21:
Maybe, but I don't repair my meshes in the slicer. I want to have control so I repair them myself, manually.

Download this that I attached. And follow my instructions.

https://github.com/prusa3d/PrusaSlicer/files/9565227/cube-broken-and-fixed.zip

DragonFinlay, 2022-09-22:
Yup, I could reproduce it with the new files. It should hopefully just be a simple UI fix.

markuskruse, 2023-08-31:
This has been fixed now. Maybe some other change fixed it or you forgot to link this issue?

closed 2023-08-31 by markuskruse