prusa3d/PrusaSlicer #8871: G-Code viewer is very slow and loads GPU heavily on mouse move

gudvinr, 2022-09-13:
### Description of the bug

Problem is that whenever I move mouse inside opengl context (both gcode viewer and parts viewer) PS loads GPU and it starts to stutter and skip frames.

If I move mouse over widgets everything is fine.

G-Code viewer seem to be affected more because it stutters significantly and GPU is loaded at 100%. Parts viewer doesn't seem to be affected visually but still I see heavy GPU load.

I started noticing that after update to 2.5. I did not check 2.4.2 but I would notice this significant performance drop. However, before I used native PS provided by ArchLinux maintainers (2.4) but now I use AppImage version (2.5).

System Information:
```
Operating System: Arch Linux
Kernel Version: 5.19.7-zen2-1-zen (64-bit)
Graphics Platform: Wayland
Processors: 8 × Intel® Core™ i7-8705G CPU @ 3.10GHz
Memory: 31.3 GiB of RAM
Graphics Processor: AMD VEGAM (Polaris 22, actually)
```

<details><summary>screenshots of PS with radeontop</summary>

G-Code viewer (no mouse move):
[image]

G-Code viewer (moving mouse around):
[image]

Parts viewer (moving mouse around):
[image]
</details>

### Project file & How to reproduce

1. Open PrusaSlicer
2. Move mouse around
3. Suffer
4. Slice some gcode
5. Switch to G-Code viewer
6. Suffer even more

Effect seem to be more pronounced on bigger parts so I'm attaching atrocious 900% cube which makes my PC struggling **badly**.
[xyzCalibration_cube.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9560532/xyzCalibration_cube.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 (appimage)

### Operating system

Arch Linux

### Printer model

Artillery Genius Pro

stuckis6er, 2022-09-17:
hello, same Problem here.

jerrysl, 2022-11-20:
Same here. I hear GPU coil whine when I move my mouse around the viewer. Sure enough, Task Manager shows ~65% GPU usage when doing that. Specs below:

**OS:** Windows 10 (build 19044), 64-bit edition
**Version of PrusaSlicer:** PS 2.5.0+win64
**Build:** PrusaSlicer-2.5.0+win64-202209060714
**GL version:** 4.6.0 Compatibility Profile Context 22.20.27.07.221020
**Renderer:** AMD Radeon RX 6900 XT
**GLSL version:** 4.6.0

Jan-Soustruznik, 2023-10-09:
Hello, @gudvinr, 

thank you for you report, but I think that this "normal" behaviours. There are many layers in gcode viewer for big objects. So the GPU will be more loaded - also connected the move with mouse in the scene. We have in our TODO list to make a performance fix of the G-code viewer in future. So when it will be done, it may help with better performing of the GPU load...

Thank you, have a nice day

gudvinr, 2023-10-09:
> but I think that this "normal" behaviours

In this specific case I only describe situation when you move mouse _without_ moving objects around or change viewport content in any way. It should not cause anything to be re-drawn other than cursor bitmap.

If there are other rendering issues, that's different story.

closed 2026-09-29 by prusaslicer-bot[bot]