prusa3d/PrusaSlicer #8875: File corruption after save failure

l0rdbyte, 2022-09-13:
### Description of the bug

If a file is still open in another program (in my case .ctb in Chitubox), UVTools will give an error message.  Saving again (after closing the offending file) will create a corrupt file.
Save as... a different filename seems to fix it.  Haven't tried with saving a third time.
100% reproducable.

### Project file & How to reproduce

Any file (only know of .ctb though) will work. Make sure it is open in another program (like chitubox).  Saving will then give an error.  Saving again will create a corrupt file.  I can send one of those (one was a layer cake of the original with multiple layers of 100% fill, another just had a few layers and nothing else).
Looks like it's something with Filestream that's not properly closed.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

3.6.3

### Operating system

Windows 10 x64

### Printer model

Phrozen Sonic Mini 8k

lukasmatena, 2022-09-14:
This is PrusaSlicer issue tracker. Does the issue have anything to do with PrusaSlicer?

l0rdbyte, 2022-09-14:
Okay not sure how it ended up here, I went for UVTools....  It can be removed

closed 2022-09-14 by DragonFinlay