prusa3d/PrusaSlicer #8876: Prusa slicer 2.5.0 failed print

gorsek1, 2022-09-14:
### Description of the bug

With prusa slicer 2.4.2 this printed without errors.
But now the print fails. Will upload a picture of the failed print later.

### Project file & How to reproduce

[53 Game Over.zip](https://github.com/prusa3d/PrusaSlicer/files/9563117/53.Game.Over.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 10

### Printer model

Prusa Mk3s

lukasmatena, 2022-09-14:
> Will upload a picture of the failed print later.

Please, do. We tested your 3MF on our MK3S without issues:

[image]

gorsek1, 2022-09-14:
Those on the left are failed. made with default 2.5.0 prusa settings.
Prusament PLA.
[image]

lukasmatena, 2023-02-08:
> I am gonna reslice them again with 2.4.2 and see how it goes.

Has this moved anywhere? Given that the 3MF you provided seems to be fine, we will close the issue shortly unless there is more info.

rtyr, 2023-05-17:
No response.

closed 2023-05-17 by rtyr