prusa3d/PrusaSlicer #8877: Detect thin walls is disabled, but no visual indication

rbronco21, 2022-09-14:
### Description of the bug

Problem - Detect thin walls is disabled because Arachne is enabled, but it is still looks active. It does not enable when clicked, but there is no indication why it cannot be enabled until you look it up in the manual.
Expected - Detect thin walls checkbox shown as disabled and a tooltip that explains why.

### Project file & How to reproduce

Try to enable Detect thin walls while Arachne is enabled...

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.50+win64

### Operating system

Windows 11

### Printer model

Prusa Mini

kubispe1, 2022-09-14:
You probably mean this :
[image]

lukasmatena, 2023-10-21:
Fixed in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-21 by lukasmatena