prusa3d/PrusaSlicer #8879: Infill caused by solid infill threshold area does not seem to use infill settings

edwardhorsford, 2022-09-14:
### Description of the bug

Discovered whilst attempting to recreate clear prints following [these instructions](https://www.printables.com/model/15310-how-to-print-glass).

My print file is using zero top and bottom layers, 100% infill, and aligned rectilinear pattern - this is to get *all* layers having layer lines going in the same direction.

You can see this in this gif:
[image]

If I set `solid infill threshold area` to a non zero value, it seems the produced infill uses standard alternating infill, not my selected infill settings:
[image]

Between these two gifs, all that's changed has been to set `solid infill threshold angle` to 100mm^2.

If I change my infill to Hilbert Curve, the difference is more pronounced:
[image]

## What I expected:

I would expect two things:

* That there would be no effect in practice because infill is already set to 100%
* That the infill caused by the setting would use the *same* infill settings as my project (if possible)

I suspect the cause here is that many infill types don't work at 100% infill, so this setting uses a default infill style. However I think where the project is *already* using a solid infill, it could respect the existing choice (or not do anything at all since the area is already set to 100% infill?)

## Impact

For my use case, there is no impact. I can simply disable the setting as I'm already using 100% infill. However it would probably be nice if this setting used the project infill setting if that setting was one that worked for 100% infill.

### Project file & How to reproduce

[solid-infill-threshold-area-bug.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9568848/solid-infill-threshold-area-bug.3mf.zip)

Open attached file and check the sliced file - for the small section, the infill angle alternates - it's clearly not using alined rectilinear.

If you disable solid infill threshold area, you'll see aligned rectilinear is used correctly.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

MacOS

### Printer model

Prusa Mini+

DragonFinlay, 2022-09-21:
I think there might be a misunderstanding here. The infill you're printing (dark red) is not "solid infill", it's just "(internal) infill" set to 100%. "Solid infill" refers to the lines that are printed inside top and bottom layers.

So if you enable a threshold for solid infill, it will suddenly generate top/bottom layers which are printed cross-hatched (purple).

edwardhorsford, 2022-09-21:
@DragonFinlay Thank you for looking at this. I may have misused some names, but I think the original issue remains - the infill type used by solid infill threshold area does not match the type selected for `Fill pattern`. Nor does it match the type selected for `Top fill pattern`. It doesn't seem possible to set the fill pattern used by `solid infill threshold area`.

I can see the issue extends to all `solid infill` - eg infill printed in top layers that are *not* the uppermost layer - the fill type cannot be set. It seems to always use monotonic.

This might well not be a bug - but I found it unexpected - and it possibly is a missing setting (though I appreciate not wanting too many settings). It's unexpected because I can set all three fill pattern settings and yet *still* have a different fill pattern get used.

[image]

[image]

DragonFinlay, 2022-09-21:
I understand what you mean now. You're right, it's not possible to set the fill pattern for these layers. Also, if top/bottom layers are set to 0, the fields for top/bottom infill pattern will be greyed out.

We should be able to do something about this, I'll address it with the rest of the team. Thanks for the explanation!

tjstansell, 2024-02-18:
This appears to still be the case even on PrusaSlicer 2.7.1.  I'm running into this trying to print a clear print using vase mode.  If I select 4 solid bottom layers, the first layer uses the bottom infill pattern, but all other solid layers (2-4) use rectilinear and I cannot adjust it.  Any chance this is being worked on?

closed 2026-09-29 by prusaslicer-bot[bot]