prusa3d/PrusaSlicer #8881: Print speed does not affect print time estimate

Wathiant2, 2022-09-14:
### Description of the bug

I had a 4h35min print with 2h1m being spent on infills. I then adjusted both the Infill and Solid infill parameters in the Print settings from 80 to 100. I expected the time to go down a bit, but I get the exact same time estimate.

I am printing the result to see if the actual speed was affected and it's just the reported estimate that's incorrect, but either way it should be fixed or made more clear that this setting does not actually reduce printing time (which would seem very strange to me).

### Project file & How to reproduce

I think this is trivial to reproduce based on the description.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S+

rtyr, 2022-09-14:
Please provide your 3MF project

Wathiant2, 2022-09-14:
[wall_mount_spool_holder.zip](https://github.com/prusa3d/PrusaSlicer/files/9569431/wall_mount_spool_holder.zip)

rtyr, 2022-09-14:
The speed is limited by maximum volumetric speed set in the filament profile, so increasing the speed in print settings has no effect. 

Expected behavior, I am closing the issue.

closed 2022-09-14 by rtyr