prusa3d/PrusaSlicer #8882: Full flow at bridge infill turnaround

espr14, 2022-09-14:
**Is your feature request related to a problem? Please describe.**
Thin bridge lines do not stick to the rest of the object.

**Describe the solution you'd like**
When the direction of bridge print changes it should stick to the object. Extrusion should be at 100%.

**Describe how it would work**
Change the marked G1 extrusions to 100% flow.

**Describe alternatives you've considered**
Keep the bridge extrusion at 100% all the time. This creates ugly bridges.

**Additional context**
[image]
[bridge turnaround.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9569428/bridge.turnaround.3mf.zip)

kubispe1, 2022-09-15:
If I understand it, you want to hard code 100% for bridge flow ratio? I think much useful could be some kind of lower limit. But in same time there are maybe users who use low bridge flow ratio for some experimental purpose and so on....

espr14, 2022-09-15:
No. I use low bridge ratio because it creates nicer bridges (fast cooling, smaller sagging).
Bridge infill is printed according to this scenario:
1. print bridge line (forwards)
2. print sideways
3. print bridge line (backwards)
4. print sideways
5. print bridge line (forwards)
.
.
.

The sideways G1 code could be printed in 100% or more. This leads makes the turnaround stick to the object. The only issue is with bowden extruders because it changes nozzle pressure. In that case the speed can be decreased to achieve thicker line with the same filament flow.

closed 2026-09-29 by prusaslicer-bot[bot]