prusa3d/PrusaSlicer #8883: gcode reading

P1RA7A, 2022-09-14:
### Description of the bug

Hello, With the new update, in my Ender 3 S1 Pro, all the G-Code projects that they carry out, the printer does not recognize the G-Code, The printer stays in automatic Bed Leveling mode and after 5 minutes it comes out error on the screen.
I've restarted the program, completely uninstalled it and reinstalled it, and the problems continue.
The same projects, I have passed it to other programs and everything works fine.

Hola, Con  la nueva actualización , en mi Ender 3 S1 Pro, todos los proyectos de G-Codes que realiza, la impresora no reconoce el G-Code, La impresora se queda en modo Nivelación de cama en automáticos y a los 5 minutos me sale error en la pantalla.
He reiniciado el programa, lo he desinstalado completo y vuelto a instalar, y sigue con le problemas.
Los mismos proyectos, lo he pasado a otros programas y todo funciona bien.

### Project file & How to reproduce

[SOPORTE_1h42m_0,20mm_210C_PLA_ENDER3S1PRO.zip](https://github.com/prusa3d/PrusaSlicer/files/9569749/SOPORTE_1h42m_0.20mm_210C_PLA_ENDER3S1PRO.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 10 

### Printer model

Creality Ender 3 S1 Pro

rtyr, 2022-09-14:
Could you also attach some g-code which works for you?

FYI @pmjdebruijn

P1RA7A, 2022-09-14:
In the message, I leave you a compressed Gcode that has been valid for me in another program.

I'm going to send you a more basic Gcode, in case you're going to print it.

pmjdebruijn, 2022-09-14:
It might be a firmware bug... Do you have your S1 Pro on the latest firmware? And if so, which version exactly?

Also, if you remove G29 from the start_gcode in the printer settings, does the problem then disappear for you?

rtyr, 2022-09-14:
It is not possible to upload files via e-mail, you have to add it via web interface.

P1RA7A, 2022-09-14:
Thank you very much, you are an artist, just removing the G29 from the configuration, it prints the pieces again, Thank you
XD

rtyr, 2022-09-14:
From Creality slicer (no G29)

gudvinr, 2022-09-15:
Most likely it has `RESTORE_LEVELING_AFTER_G28` turned on in firmware

pmjdebruijn, 2022-09-15:
@gudvinr yeah I figured as much.

@P1RA7A you didn't tell me your printer's firmware version yet...

P1RA7A, 2022-09-15:
Model: Ender 3 S1 PRO
F/W : 2.0.8 15F4
DISPLAY: UI20.2
H/W: CR-FDM-v2.5.S1_1

pmjdebruijn, 2022-09-15:
2.0.8.2-SW-V2.0.8.13-MP-20211214 seems to be the newest, would you mind retesting with G29 enabled and the latest printer firmware?

P1RA7A, 2022-09-15:
Good afternoon, just now I was trying to print a part, with the G29 auto bed leveling activated, and I still have the error. I have to delete it from the Gcodes configuration and save the profile, to be able to print, last night I was able to make a small piece without problems, since I was able to disable it.
There must be some incompatibility with the latest version of my printer and Prusa slicer, the old version of Prusa did not give me any problems.
Could it be because of the Laser Marlin that Creality took out?
It could be, I don't know.
***@***.***

pmjdebruijn, 2022-09-15:
@P1RA7A What happened isn't a mystery, but G29 failing this badly is most likely a firmware bug, which is why I'd like to you to try the latest firmware.

rtyr, 2022-09-21:
Fixed with 0.2.2 configuration update.

closed 2022-09-21 by rtyr