prusa3d/PrusaSlicer #8884: Brückenwinkel

Herby0107, 2022-09-14:
### Description of the bug

Der Winkelwert wird nict verwendet!
Habe 180° eingestellt, aber wird mit 45° verwendet.

### Project file & How to reproduce

[AD O 02.zip](https://github.com/prusa3d/PrusaSlicer/files/9570150/AD.O.02.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Win 10

### Printer model

Eigenbau

neophyl, 2022-09-15:
Thats not a project file you attached, that's just the stl file.  To debug slicer issues the devs generally need the full saved project file as this contains all the settings you are using.
You need to use File>Save Project As to save the 3mf file.  Then zip and attach that.

Given I have just tried altering the bridging angle with your attached model and it changes fine and as expected (with 2.5 on a windows 10 setup) then it suggests that the problem is with the settings you are using.  So please attach a project file so we can see them.

Herby0107, 2022-09-15:
[AD O 02.zip](https://github.com/prusa3d/PrusaSlicer/files/9573324/AD.O.02.zip)

Herby0107, 2022-09-15:
Hier das Projektfile

foreachthing, 2022-09-15:
Also, wenn ich den Winkel auf 180° einstelle, dann erhalte ich auch 180°??
[image]

Oder ich verstehe Dein Problem nicht ganz?!

foreachthing, 2022-09-15:
Das funktioniert wirklich nur, wenn der "Füllwinkel" auch auf 180° eingestellt wird.
Fiel mir beim letzten Teil auch auf ... :raised_eyebrow:

neophyl, 2022-09-15:
The problem is that in the supplied project its NOT being treated as bridging.  SO changing the bridging angle does nothing.  Notice that the layer that is normally bridging should be keyed as light blue, its not.  If I change the print profile to one of the normal ones it does then bridge and uses the set angle.

I haven't yet figured out what setting or combination of settings is causing it to not be treated as bridged though.

neophyl, 2022-09-15:
Found it.  Its the supports contact distance.  Its set to zero which is for soluble support.  When using soluble support bridging is disabled.  Ergo no bridges and bridging angle doesnt do anything.

Herby0107, 2022-09-15:
Alles Klar, wenn das nicht als Überbrückung erkannt wird geht es nur mit
dem Füllwinkel.
Danke
Solte aber als Überbrückung erkannt werden.

closed 2022-09-15 by Herby0107