prusa3d/PrusaSlicer #8885: Estimated printing time

BerlinLion, 2022-09-15:
### Description of the bug

The software showed a printing time of 15h and 4 minutes, but now after 19h and 22 minutes, it is still printing. This is since the newly installed update. Before it was actually precise on the minute.

### Project file & How to reproduce

Import a stl file, slice it, export the G-Code and print it out.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

Anycubic Vyper

zmeygeruch, 2022-09-15:
> ### Description of the bug
> The software showed a printing time of 15h and 4 minutes, but now after 19h and 22 minutes, it is still printing.

Hi. Try to check you machine settings in slicer and compare it with real speeds settings in your printer.
[image]

rtyr, 2022-09-16:
I believe it was already answered by @zmeygeruch. You have to set machine limits properly. The first setting "how to apply limits" is also important, because your FW can use lower limits by default.

I am closing this issue. If you still think, there is some issue on the PrusaSlicer side, please provide 3MF.

closed 2022-09-16 by rtyr

BerlinLion, 2022-09-24:
The point is, that I use the simple settings mode and I had to activate the
expert settings mode to see them at all. I only updated the Prusa software
and before the update it made an astonishing precise prediction regarding
the print time. But now the difference is larger than 30% and as I said, I
changed nothing regarding the printer settings. I use the same printer and
everything remains the same since the update. However, I'll have a look on
how to fix it or I'll find the exact percentage to make the prediction
better.