prusa3d/PrusaSlicer #8886: 2.5.0 Regression - Empty Layers In Thin Pointy Tower

Thynix, 2022-09-15:
### Description of the bug

I'd expect all layers, especially in a tower with no overhangs, to be printed on top of previous layers. 2.5.0 generates empty layers and floating layers, and 2.4.0 does not:

[image]
[image]

Given this model:

[image]

### Project file & How to reproduce

Here's the project file from the 2.5.0 screenshot: [ctrlv_3D_test_v3.zip](https://github.com/prusa3d/PrusaSlicer/files/9572031/ctrlv_3D_test_v3.zip)

This is from slicing [Test your 3D printer! v3](https://www.thingiverse.com/thing:1363023/files) by ctrlV.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

Prusa MINI+

zmeygeruch, 2022-09-15:
Hi. I Think, this happens, because "detect thin walls" was disabled permanently in 2.5.0 :(
[image]

neophyl, 2022-09-15:
Its caused by the arachne slicing engine now being used as default.  If you change the slicing engine back to classic you get the old behaviour.  You also get detect thin walls enabled again.  The point with arachne is that it varies extrusion widths as it goes along and so makes detect thin walls pointless.

There have been several of these issues raised which are all basically the same thing, people being caught out by the change to arachne, even though its in the release notes.  It just means you have to tune the settings for your models differently or switch back to tried and true if the new engine isn't to your liking.

kubispe1, 2022-09-15:
Thank you guys. 
It is duplicate of #8730 and many more :D, so I close it.

closed 2022-09-15 by kubispe1