prusa3d/PrusaSlicer #8888: Brim modifier interfering with other brim modifiers

airiclenz, 2022-09-15:
### Description of the bug

When adding the brim modifier to multiple objects (stl), the resulting brim looks to be affected by the near by brims of other objects - even though they are not touching and should be 100% separate from each other:

[image]

### Project file & How to reproduce

The problem seems to appear when the brim borders get close to each other.
Reducing the brim with - and thus increasing the distance to the next brim - seems to fix the issue.
moving the object further away also solves the issue.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Mac 2.5.0

### Operating system

macOS 11.7 Big Sur

### Printer model

Voron 2.4 with Afterburner

Snuff1eupagus, 2022-09-19:
That is normal behavior for Brims and is desirable. if you allow them to grow larger they will reach for each other and attempt to split the difference between the two separate sections of Brim. It works out great.

DragonFinlay, 2022-09-22:
@Snuff1eupagus I understand what you mean, but I think that this particular example is actually not how brim should behave.

It happens because two objects have a different brim width, and the bigger brim accounts for the smaller brim even though the two brims are not touching each other at all.

[image]

I'll discuss this with the team, but I imagine changing the algorithm that generates the brim might be time-consuming and this should not interfere with the printability of the parts for now.

Snuff1eupagus, 2022-09-23:
No argument here, but I think if you really understood what can be done with Brims your response would be different.

Regards

DragonFinlay, 2022-09-23:
Please feel free to explain or give me an example, I'm always open to suggestions! :)

airiclenz, 2023-01-20:
The issue with my original example is, that very short sections are printed. This might cause them to not stick properly to the bed and will thus cause problems when these loose parts are dragged over the other parts of the first layer.

I would love to get some additional explanations as well.

Snuff1eupagus, 2024-04-26:
Perhaps this will enlighten you.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=4s

Enjoy

closed 2026-09-29 by prusaslicer-bot[bot]