prusa3d/PrusaSlicer #8890: Ender V2 Print and Filament settings disappeared.

darthloser, 2022-09-16:
### Description of the bug

After upgrading to PrusaSlicer 2.5.0, the Print and Filament settings for my Ender 3 V2 preset have disappeared, there is only "-default-"

### Project file & How to reproduce

Issue occurred on upgrade, unsure how to replicate.
[PrusaSlicer problems.zip](https://github.com/prusa3d/PrusaSlicer/files/9579664/PrusaSlicer.problems.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Mk3s

rtyr, 2022-09-16:
What was your previous PrusaSlicer version? Was it some alpha/beta release? 

> the Print and Filament settings for my Ender 3 V2 preset have disappeared

Your custom settings or bundled system Ender 3 V2 presets?

pmjdebruijn, 2022-09-16:
Thanks for reporting, and apologies, this is likely due to a capitalization typo...

rtyr, 2022-09-21:
Should be fixed with 0.2.2 configuration update. Closing.

closed 2022-09-21 by rtyr