prusa3d/PrusaSlicer #8891: Online profile to sync settings/config between machines

Combinacijus, 2022-09-16:
**Is your feature request related to a problem? Please describe.**
I use multiple machines for slicing and because filament profiles are changing constantly it's hard to keep settings synced between machines by manually exporting and importing configs.

**Describe the solution you'd like**
I would like some kind of online/cloud based automatic syncing. Maybe config could be synced over Printables.com or Google Drive or etc. 

**Describe how it would work**
- Download PrusaSlicer 
- Login to online account (or skip if not interested)
- Do the config sync automatically or give a warning on merge conflict
- Enjoy same configs on multiple machines

**Additional context**
Additionally it would act as backup in case your machine gets corrupted etc. No need to manually do config backups after every change.

rtyr, 2022-09-16:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/6780.

Moerty, 2022-09-16:
#4898

rtyr, 2022-09-20:
Closed as a duplicate.

closed 2022-09-20 by rtyr