prusa3d/PrusaSlicer #8894: Adding Extruder Z offset to individual extruders

DegreelessEngineering, 2022-09-16:
**Is your feature request related to a problem? Please describe.**
When using the multiple extruder function, there is an x and y offset setting, but no Z offset setting. This makes PrusaSlicer incompatible with printers like the Raise3D Pro3 printers that have an automated right extruder (T1) that extends and retracts when selected/deselected. Without this z offset adjustment, the right extruder crashes into print/bed. I'm assuming it's due to the slicer referencing the left nozzle z-probe offset instead of the right nozzle z-probe offset. I've tried everything I know or could find and nothing works.

**Describe the solution you'd like**
Add z offset adjustment under each extruder

**Describe how it would work**
When the tool is selected via the g-code, it detects the x and y offsets and adjusts accordingly. To this same effect, but for the z-offset. This would work for tool changer printers (like the upcoming XL) as well because every tool/mount is going to be slightly different.

**Describe alternatives you've considered**
Maybe I don't understand how it works, but I've tried using macro within the custom gcode, but Prusaslicer did not understand it. It also doesn't let you indent, which from what I understand is required when using macro. Instead it just puts you in the next window. I've tried z-hops in the tool change gcode, but for some reason when I do this it puts a random layer of the tool being selected deep within the wipe tower (like many layers down). However, the rest of the layers are exactly where they need to be. This is very weird and leads the right nozzle to drag through/crash into the wipe tower because it's trying to reach that layer.

**Additional context**
I've already developed presets for the left nozzle only and right nozzle only, but when printing with both, it does crazy stuff. There are TONS of these printers out there and everyone that uses them prefers PrusaSlicer.

rtyr, 2022-09-19:
I don't understand how would z-offset help if the nozzle extends/retracts. This should be handled completely by the firmware. My guess is that your profile is probably missing some Raise3D specific tool change commands.

rtyr, 2022-09-19:
So I took a quick look into the Ideamaker (original Raise3D slicer):

- Ideamaker supports only X,Y offset and it is even not being used for g-code generation, only for print volume check in the preview. X,Y offset for g-code is set to 0,0.

- There is no Z-offset for individual tools in the g-code generated by Ideamaker for Raise3D Pro3 (and I still don't understand how could it help for such setup).

- X,Y offsets in the PS are needed only if FW is not properly configured. Raise3D firmware is likely correctly configured in this regard.

TomaszSzt, 2022-09-20:
>     * X,Y offsets in the PS are needed only if FW is not properly configured (...)

You got the point. Why not allow to help those who have Z offset incorrectly configured?

DegreelessEngineering, 2022-09-20:
- Makes sense. I was looking at the virtual offsets.

- If the PrusaSlicer x and y offsets under Extruder 2 are left at 0 and 0, the x and y are going to the correct position without entering them manually. However, I'm still experiencing a weird issue when the wipe tower is enabled:
1. Extruder 1 completes its portion of the wipe tower and part being printed
2. Extruder 2 is selected and extends nozzle, but z axis does not change.
3. Extruder 2 nozzle drags through the wipe tower during its first wipe layer
4. Extruder 2 then moves to actual part being printed and THEN adjusts the Z Axis to offset the nozzle extension.
5. Extruder 2 then moves back to the wipe tower and the z axis is now at the correct height for the wipe tower and no longer dragging though it.

For some reason when the right extruder is selected it always drags through the wipe tower on its first wipe tower layer. I thought me removing my manual entry of the x and y would have fixed that, but it's still crashing into the wipe tower on the first layer. I then completely removed the following "Tool Change Gcode" I had in place and that solved the crashing issue:

G92 E0
T[next_extruder] P0 ; select tool

Now when the right extruder is selected, the z-axis adjusts with the nozzle extension BUT the nozzle is no longer in line with the wipe tower. It now travels to the left and hits the x-axis max and prints a blobby line back and forth. I'm going to try reinserting the tool change gcode MINUS the P0 because I wonder if that's preventing firmware macros from running like they should. Not sure.

closed 2022-09-20 by rtyr

h8nor, 2022-10-31:
Is a similar problem with the 3D Systems printers. The S3D slicer inserts `M101` and `M103` g-codes, and uses relative coordinates with negative nums. I suspect that because of such features, files in the [PrusaSlicer G-Code Viewer](https://help.prusa3d.com/article/creating-profiles-for-different-nozzles) are not read.
Users indicated for the S3D printers the following g-codes should not be used. Configuration:
```ini
[print:NAME]
gcode_substitutions = (?:; estimated printing time \(normal mode\) = (\s*));"^Time: ${1}";r;PRINTING_TIME_NOT_WORKED
gcode_substitutions = ^(G90|M(82|102|404)|T[012]);";; ${1}";r;STRIP
[printer:NAME]
gcode_flavor = marlin
before_layer_gcode = ;BEFORE_LAYER_CHANGE\nG92 E0.0\n;LAYER:{layer_num} Z{digits(layer_z, 0, 3)}
use_relative_e_distances = 1
```
_[The configuration is downloading with continuous text](https://github.com/prusa3d/PrusaSlicer/wiki/Vendor-bundles-and-updating-process) and requires manual revision for publication in the repo._