prusa3d/PrusaSlicer #8895: Uneven/Ugly layers, when using arachne perimeter generator.

teodortomas, 2022-09-17:
### Description of the bug

_**What is the problem?**_

When perimeter generator is set to Arachne, then wierd ugly layers appear. The problem is visible on all models (not at the same layer). When switched back to classic, prints look good again.

_**What did you expect?**_

I expected the Arachne perimeter generator to work.

(I found one person with the same problem)
[image]

### Project file & How to reproduce

[test_object.zip](https://github.com/prusa3d/PrusaSlicer/files/9591657/test_object.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 11

### Printer model

Ender 3 V2 with E3D V6 hotend and Direct feed BMG

pa0los, 2022-09-18:
I have observed the same issue with latest SuperSlicer:
https://github.com/supermerill/SuperSlicer/issues/3236

kubispe1, 2022-09-19:
It seems to me that the inner shape is a polygone curve and the outer perimetr looks more circular. So there are thickness transitions. Than Arachne is pushed to compensate it.  This could, maybe, cause these artifacts on the facade surface as you reported.

[image]
[image]

alse I think, that here is duplicate issue (but I was not able to reproduce it} :
#8524

teodortomas, 2022-09-19:
> It seems to me that the inner shape is a polygone curve and the outer perimetr looks more circular. So there are thickness transitions. Than Arachne is pushed to compensate it.  This could, maybe, [...]

You are right about the round outer exterior, but it happens to me on other models too (not vase mode prints)

DragonFinlay, 2022-09-23:
As observed by @kubispe1 here and me in #8524, this is definitely a result of the level of detail these 3D models have. If the subdivisions of the inner walls don't match the subdivisions of the outer walls, Arachne will automatically compensate for thinner and thicker parts. Of course, these subdivisions are difficult to control without advanced modeling software.

I closed the other issue but we might re-open it to implement some sort of tolerance in PrusaSlicer. Please give us some time to take a closer look.

closed 2026-09-29 by prusaslicer-bot[bot]