prusa3d/PrusaSlicer #8896: Inconsistent bridge infill behavior depending on perimeter count

th33xitus, 2022-09-17:
### Description of the bug

For a drafting profile i regularly use rather sparse infill. In particular 15% adaptive cubic infill. Usually i also pick 3 perimeters by default. Now the issue:
While using that amount of infill, bridge infill often can't anchor to the previous layer as there is just no infill below it. But what i don't understand is, why the bridge infill is not getting anchored to the inner perimeter. That's a reason why things like this happen: 
[image]

Now the inconsistency i mentioned. If i increase the perimeter count to 4, the gap gets narrower:
[image]

If i further increase the perimeter count to 5, the gap (at least the one in the upper and lower part of the screenshot) is closed and i would even guess, the bridge infill will attach to the inner perimeter when printed:
[image]

I don't quite understand the mechanics behind the bridge infill, and why it behaves like this. I would rather prefer the bridge infill to anchor into the perimeter, because with no infill below it, it either droops down into the part, or partially curls up, causing the nozzle to drag over it during the following layer.

I observe this problem especially in situations where further printing is done on a top solid infill later in the process:
[image]

### Project file & How to reproduce

Reproduction is easy, open the project and change the perimeters from 3 to 4 to 5 and check layer 14 in between the changes. You will observe an inconsistent behavior of the bridge infill throughout that layer. In one third of the part, the bridge infill gets anchored to a short strip of solid infill, in another part its "anchored" to some internal infill (doesn't look correct to me either, looks very weak), and in some places, bridge infill is basically printed in thin air, with nothing to attach to, resulting in it to curl up/droop down.
[image]

Project file and step file attached:
[project_file.zip](https://github.com/prusa3d/PrusaSlicer/files/9591708/project_file.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

V-Minion

DragonFinlay, 2022-09-17:
Bridge infill looks at the cavity, not the body below. It takes the bottom face of the cavity, expands it by a few millimeters, and generates the bridge infill from that. It's independent of the outer walls around it.

Take this scenario: Your main body is big and the cavity small. Printing a full bridge infill layer just to attach to the outer walls would be a waste of time and material:

[image]

What could work well for you is the new Lightning infill. It gives you much more support for the bridge infill while still saving infill material (in this case with the pictures below, grid infill uses 51g of material while lightning infill uses 37g).

[image]
[image]

th33xitus, 2022-09-17:
> Take this scenario: Your main body is big and the cavity small. Printing a full bridge infill layer just to attach to the outer walls would be a waste of time and material

Good point, makes total sense. It would indeed be a waste of material.

In your example there is also an extrem case of bridge infill that is unsupported and from the looks of it would be pretty much unprintable. A feature for expanding the bridge infill area would be awesome. In your case for example, if you could say "expand until nearest" it could expand it like this and maybe anchor the bridge infill to the next near internal grid infill:
[image]
In that case you would anchor the bridge infill all around, create a good foundation for the upcoming layers but at the same time would not waste that much material on it. And in cases like mine it could just expand the area up to the inner most perimeter if there is no other infill it can anchor to.

Might be worth considering similar options to the "Length of the infill anchor" and "Maximum length of the infill anchor" we have for the infill settings. Just applied to the bridge infill area and anchoring technique.

Thanks for your response!

DragonFinlay, 2022-09-17:
I really like your idea with "expand until nearest", that would indeed make a lot of sense. I'll have that forwarded to the rest of the team!

DragonFinlay, 2022-09-17:
We found this suggestion in an old issue (#569), it seems to have been on our to-do list for quite a while...
Closing this one as a duplicate.

closed 2022-09-17 by DragonFinlay