prusa3d/PrusaSlicer #8898: Slice Analasys across different planes

mvasilakis, 2022-09-17:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to see across section of the slice in the vertical direction. Similar to what Fusion 360 analyses does. 

**Describe the solution you'd like**
A slider to move across the X plane to view a cross section of the slice. (As opposed to the available Z cross-section which shows the layer slice.)

**Describe how it would work**
I'd expect it to work like it does in Fusion360 where I can slide across the X axis and see inside the sliced model.

**Describe alternatives you've considered**
I don't really know of another way to do this.

**Additional context**

[image]

DragonFinlay, 2022-09-17:
This is already possible with a little workaround:

1. Import your model into PrusaSlicer.
2. Move the camera to the position parallel to the plane you want to cut (e.g. using the numbers 1-6 on the keyboard).
3. Open either the paint-on-supports or seam-painting feature in the left sidebar.
4. Use "Clipping of view" to initiate the cross-section view.
5. Feel free to move the camera around as the cutting plane is now locked to the plane chosen in step 2.

Maybe something more sophisticated will be added in the future, but for now, I think this should be sufficient.
Closing this issue.

closed 2022-09-17 by DragonFinlay

mvasilakis, 2022-09-18:
I gave that a shot. It's similar to the Fusion360 analysis however it doesn't visualize the infill and layers. After a part is sliced, depending on the infill percentages and wall thickness sections may become too hollow or unnecessarily thick. If we can slide in and out like you show and see the infill it would help allot for models that we are trying to optimize. 
I get that it's probably not going to be a priority any time soon but please keep it in mind. For functional parts it helps allot.

rtyr, 2022-09-19:
Same as https://github.com/prusa3d/PrusaSlicer/issues/5973.

DragonFinlay, 2022-09-27:
As such, we can keep it closed. But I agree, something like this would be useful.