prusa3d/PrusaSlicer #8904: 2.5.0 Uninstall Old Version results in series of errors after uninstall

seanmcne, 2022-09-18:
### Description of the bug

2.5.0 installation successfully uninstalls old version but then fails to continue installing 2.5 

Have reproduced on 4 PC's 

### Project file & How to reproduce

Install 2.4.x
Install 2.5.0 and select dialog option to uninstall the old version. 
Old version successfully uninstalls however, 2.5.0 then errors out 

[image]
Press retry and then get this: 
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 RTM

### Operating system

Windows11 and Windows 10

### Printer model

MK3S+

seanmcne, 2022-09-18:
FYI workaround is to Abort 2.5.0 installation and restart the installation process for 2.5.0

Janys, 2022-09-19:
Hello, I'm responsible for the installer and I'll investigate the issue. Can you please clarify one thing?

You mentioned you "Have reproduced on 4 PC's"

Did you use always the same file (e.g., copied onto a USB flash drive) or did you download the installer separately on each of those computers? Did you download it from the web or did you use the auto-updater? Because the FDI error usually means that the file is corrupted

Thank you

seanmcne, 2022-09-20:
Hi Janys, thanks for replying! 
The file I used to install was downloaded (each time) from the prusa website directly (each time).  If I stop the installation and restart it (after it successfully uninstalls 2.4.x) it continues to run successfully.  The FDI error only comes after the access denied error - I would say it should be pretty easy to reproduce if you have 2.4.x installed then download the exe for 2.5.0 then install, you'd get prompted to remove 2.4.x, after that works, the 2.5.0 installation resumes and that's when I get the error.

motykap, 2022-09-20:
I have experienced the same, exactly the same precedence. After installation was completed (despite all these errors) slicer would not start and error message pops up - PrusaSlicer config file corrupted. I did another download and installation but same result.

closed 2026-09-29 by prusaslicer-bot[bot]