prusa3d/PrusaSlicer #8906: Crash after deleting preset assigned to physical printer

the-ress, 2022-09-18:
### Description of the bug

After deleting preset assigned to physical printer, PrusaSlicer crashes due to Access violation (`c0000005`) at `PrusaSlicer+0x0000000000069132`.

I couldn't find debug symbols so I didn't investigate further. (Please publish debug symbols with each released version.)

After restarting the PrusaSlicer, the physical printer is missing, and its `ini` file only contains:
```
# generated by PrusaSlicer 2.5.0+win64 on 2022-09-18 at 22:03:20 UTC
```

[image]

### Project file & How to reproduce

1. Have a physical printer that has an user preset and one or more other presets assigned
2. Go to Printer Settings tab
3. Delete the user preset
4. In the dropdown, select one of the presets in the Physical Printers section
5. Press the cogwheel icon (Edit physical printer)
6. PrusaSlicer crashes

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 11

### Printer model

Prusa MK3S+

kubispe1, 2022-09-19:
Hello, if it is not problem, can you send us printer bundle?
[image]

lukasmatena, 2023-11-12:
Can you please provide the data? Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]