prusa3d/PrusaSlicer #8907: Automatic Model Splitting

HalfPastMidnight, 2022-09-19:
When a model is too large for the build plate, there should be a feature that can automatically split the model into multiple parts to fit the size of the build volume.

Ideally it would just be a singular button that splits the model into the largest possible pieces in comparison to the build volume. 

This could be as simple as taking the total x, y and z measurements of a part and splitting it using the existing manual splitting feature, then open new windows with the other split parts ready to be sliced. I think this would be a simple math function. I.e. if diameter x of part = 300mm, but the diameter x of the build volume is 250mm, the part would be split in half so that 2x 150mm parts would be printed instead. Each of these 150mm parts would be opened in a new windows. 
(Joinery of the parts would be difficult after printing so maybe some sort of location holes and posts could also be added during the splitting process, although I am sure this would not be a trivial problem. I predominately print in ABS, so ABS juice would be used in my case and location holes and posts are not necessary)

The alternative to this solution is that I just adjust my workflow and allow time to split the models manually, although I think this feature would not be too difficult to implement.

stevrgrs, 2024-03-10:
100%  I was just trying to do this and its a pain.  Have you found any solutions yet?

Heliosmaster, 2024-12-18:
I second this. But i understand it will be very difficult

closed 2026-09-29 by prusaslicer-bot[bot]