prusa3d/PrusaSlicer #8908: Horizontal gap in outside parameter that moves each print

jvcjbl, 2022-09-19:
### Description of the bug

Ever since the Arachne update (ender 3 v2) whenever I print anything (tried re-slicing already) I'll get a random section where there is a gap in the outside parameter of the wall. I've tried lowering the wall count, raising the wall count, printing outside parameters first nothing eliminates it. I even tried switching back to classic and it's still there. I sliced the same model in Cura 5.1.1 and it prints without this gap, but the print takes over an hour longer so I don't use Cura anymore since for me it's always so much slower than prusaslicer. I've tried changing the seam gap to random, aligned, nearest, etc, no change. Here is a picture of the same item sliced three different times changing infill only and you can see the gap is different and moves. I even printed the same print off the same .gcode and the gap moves from print to print. Again, this has just started since the update and does not happen at all with Cura. The gap is not indicated (to my best knowledge) anywhere on the preview inside of prusaslicer, but I'd like to mention I was able to stop the print right at this gap and it's in fact printing the outer wall, but looks likes it's printed on top of the inner wall, almost like the printed is not being told to move over for the outer edge in that area.

[image]

### Project file & How to reproduce

[Ghostbusters_Penny_Bottle_Opener.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/9598187/Ghostbusters_Penny_Bottle_Opener.gcode.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+ 

### Operating system

MacOS-64

### Printer model

Ender 3 V2 with Microswiss direct drive extruder/hotend

jvcjbl, 2022-09-19:
I just sliced a completely different model and it's the same thing, starts randomly and ends randomly only in one section, the height of one layer. If it helps it seems to occur on layer 5-6 from what I can tell.

neophyl, 2022-09-19:
If you print the exact same gcode and the problem moves around then it is not the gcode but a physical problem with the printer.  If it was the code it would occur in the same place each time.

The obvious thing to check for is a partial clog.

jvcjbl, 2022-09-19:
I get that, but why does it happen on the same layer each time every print and why only with PrusaSlicer. No clogs, checked already. Sent from my iPhone

neophyl, 2022-09-20:
In which case then yes it could be gcode related.  It could still be a physical printer problem though.  I recently had a similar extrusion problem where at the same points on the same model it would fail to extrude and I would get a similar missing section.  In my case though I could hear the extruder motor and it was struggling in those sections.  I initially thought it could be 2.5 as I had just updated.  Maybe something to do with gcode per second sent, or something more esoteric.

Like you I tried a different slicer, that moved the problem to a slightly different area so , yep slicer I thought.  Just got to be the gcode.  Couldn't figure it out.  The problem seemed to get worse though so I then thought it was a faulty stepper motor.  So I replaced it.  Initially that seemed to fix the issue.  Unfortunately the problem came back a couple of days afterwards.  

Turned out to be a faulty wire connection to my extruder.  The different slicers doing different print pathing of the same models would cause the wire disconnect in slightly different areas.  Changing the stepper had obviously moved the wiring restoring function for a time.  The long winded point is though that it could still very well be a mechanical/electrical issue.

Do you know anyone else with the same printer that could try your gcode file ?  Maybe on an Ender forum.  That way you could 100% rule out any gcode produced.

jvcjbl, 2022-09-20:
No I unfortunately don't know anyone with a 3D printer, let alone someone with mine. So 100% confirmed. The "gap" doesn't move print to print. The location only changes if I reslice it, but printing the same .gcode over and over reproduces the same gap in every print. I tried taking the same file again in to Cura 5.1.1 and sliced it and printed and and zero gaps. Something in the this update is causing it as I also confirmed that switched it back to "classic" doesn't rectify the issue. I've tried loading the file directly to SD card and or sending it via Octoprint like I do everything 100% of the time and it resulted in the same gap in the same place.

jvcjbl, 2022-09-20:
So here's what I've done and here's what I've discovered. I completely deleted prusaslicer and reinstalled it fresh. Just to see I used to 100% bone stock default ender 3 v2 profiles that come in the configuration assistant. Re-sliced the file that is attached here and no more gap. I input all of the settings and the gap returns. I'm now trying to narrow down the issue, but I think I just realized that it's happened when my fans kick off. I have the fan off for the first 5 layers and if you recall I keep say it always happens on the 5-6 layer. I just changed the fan to kick on at layer 3 to see if the problem moves/goes away.

jvcjbl, 2022-09-20:
Ok so I figured it it. The file that's attached, the issue as right end/beginning of layer 5-6 when the fans kick on. This has never, ever, ever been a problem, but for whatever reason it is now. If I use the fan full speed at layer command and let it ramp up or even 1 layer there is no gap. If I allow it to turn on fully (like it has ever since I've owned this printer) at layer 6 I get a gap. Now, I have no idea whats causing this as it started with 2.5.0. I was in the middle of doing 10 consecutive prints of the same item over and over again for a customer. The update came to my system when I was about 6/10 prints completed. I did the update and replaced the file and the remaining 4 prints a gap. I took photos of each print and each one was a different color. I still have the prints in my possession as I have yet to ship then so I went back and matched each up to the original order based on the photo and sure enough after the update the gap was there. For what it's worth, Cura is set to "regular fan speed at layer 6" so it does a ramp up on it's own and has always. I don't know why it's never been a problem until now.

jvcjbl, 2022-10-06:
Not that it seems like anyone else cares or was experiencing this issue, but I resolved it. I mistakenly set the extruder minimum temp to be the exact same as what I was printing so when the fans kicked not he extruder dipped a few degrees and it stopped since I told it to not drop below the same temp I was printing at 🤦🏻‍♂️🤦🏻‍♂️🤦🏻‍♂️🤦🏻‍♂️🤦🏻‍♂️🤦🏻‍♂️.

closed 2022-10-06 by jvcjbl