prusa3d/PrusaSlicer #8915: STEP import resolution too low that it effects sliced output

5jvm0u4, 2022-09-20:
### Description of the bug

As the title implied, I found out that fillet areas need more resolution than currently set in the slicer, and this affects the final output
[image]
[image]
[image]

### Project file & How to reproduce

Import the file attached below：

[STEP resolution.zip](https://github.com/prusa3d/PrusaSlicer/files/9606041/STEP.resolution.zip)
[Plate.zip](https://github.com/prusa3d/PrusaSlicer/files/9606049/Plate.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11

### Printer model

N/A

gudvinr, 2022-09-20:
Does it affect quality of printing parts though? You don't print with meshes, plastic still flows and won't lie down with sharp angles so at some point it doesn't matter how small slicer stepping is.

foreachthing, 2022-09-20:
I use [ArcWelder](https://github.com/FormerLurker/ArcWelderPlugin) in combination with step files. That helps a lot with accuracy and cleanliness around radii - if your printer supports it.

5jvm0u4, 2025-04-05:
Finally, 2.9.2 rc-1 fix this problem

Laserology, 2025-07-28:
Bump: I still have this issue even with the new import resolution options. Even with the highest resolution set possible, i still get artifacts that are visible in the gcode that are likely to cause print defects. What is the possibility that the max resolution be increased?

Laserology, 2025-07-28:
[image]

[image]

[PhoneStand_Freecad.zip](https://github.com/user-attachments/files/21478086/PhoneStand_Freecad.zip)

Laserology, 2025-07-28:
You can also see here how it affects speed:

[image]

5jvm0u4, 2025-07-29:
I can confirm the surface artifact, but not the speed problem, glad I did not close this issue.

[image]
[image]
[image]
[image]
[image]

5jvm0u4, 2026-09-03:
Keeping this alive.