prusa3d/PrusaSlicer #8917: Remove placement guesswork, of part manipulation objects.

Kagnaw, 2022-09-20:
**Is your feature request related to a problem? Please describe.**
Placement of a 'Part Manipulation Box' for example, i want to block support in a specific spot. 
I have to guess the right position slice, move it, slice, adjust, slice and so on.
especially a problem in bigger parts if you want to adjust something in 3d space.

**Describe the solution you'd like**
Being able see and move the the modifiers/manipulations in the slice preview. Could be a toggle to show them.

**Describe how it would work**
I slice, i see something i don't like i make a part manipulation box (im still in the preview)
i can move the box too the spot i want to modify. Slice. Yell first tryyy and move on with my day.

**Describe alternatives you've considered**
Doing it by "feel" and guesswork. as i do now.

**Additional context**
Attached a random made upp example, i want to remove the solid at the end of the screw hole, so i could use longer ones or just to save print time.

[image]
[image]
[image]

neophyl, 2022-09-20:
So basically a duplicate of either https://github.com/prusa3d/PrusaSlicer/issues/3678 (which I remember as I created it)  :)  or https://github.com/prusa3d/PrusaSlicer/issues/4438 which has some nice example graphics included.

Kagnaw, 2022-09-20:
Ahh yes indeed, i see my search for similar requests failed quite badly, as there are many others.

I guess the only difference in my is that i would like to see and move modifier in the sliced preview to more easily place it right on the thing i want to modify.

kubispe1, 2024-10-29:
I am closing as a duplicate. Thanks a lot for FR

closed 2024-10-29 by kubispe1