prusa3d/PrusaSlicer #8918: Enable saving/export of variable layer height profiles for use in other models

Aero-Grip, 2022-09-20:
**Problem:**
I have a product that has many different configurations but the same overall profile and height, a handgrip for example.

Every time I reload a new model it wipes the variable layer profile and I have to manually add it back, which is time consuming and never exactly the same.

**Solution:**
I need 3 variable layer profiles to cover all configurations. I would like the ability to import/export variable layer profiles so they may be applied to other models.

**How it would work:**
Just an Import/Export function for variable layer profiles

**Things I've tried:**
Once I have a .3mf saved with the variable layer profile I want, I've tried naming a new STL model the same that was saved and reloading "from disk" in prusaslicer. The result is the variable layer profile gets wiped.

I've tried copying the file and the text from Slic3r_PE_layer_heights_profile.txt from one .3mf to another, but that doesn't seem to work either.

MNeMoNiCuZ, 2023-03-25:
+1 on this feature.
A simple way to copy/paste Variable Layer heights between objects.
At least as modifiers, but also straight from the "Variable Layer Height" panel.

There's also a bug when you have a merged object with multiple meshes, and you then delete a specific mesh. Then the variable layer painting is deleted. As if the modifier belonged to that sub-object mesh instead of the whole merged object.

MNeMoNiCuZ, 2026-09-04:
This is still not supported.