prusa3d/PrusaSlicer #8919: [Mac] Keyboard shortcut not working with Czech keyboard layout

Stooovie, 2022-09-20:
### Description of the bug

When using my default Czech keyboard layout, PS doesn't respond to keyboard shortcuts that use the numbers on the alphanumeric keyboard (cmd+1,2,3...). AFAIK Apple changed handling of these so it sends cmd+corresponding czech accented character (cmd-ě instead of cmd-2, cmd+š instead of cmd-3).

Please fix this so Czech Mac users can continue to use these keyboard shortcuts.

### Project file & How to reproduce

enable Czech keyboard layout and use cmd-# shortcuts. They won't work.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+MacOS-x64

### Operating system

MacOS 12.6

### Printer model

Ender 3

Stooovie, 2023-10-22:
Still unfixed as of 2.7.0 alpha 1

lukasmatena, 2023-10-23:
Closing as a duplicate of #9345.