prusa3d/PrusaSlicer #8920: Access Violation Slicing MK3MMU2S project

MB3Y, 2022-09-20:
### Description of the bug

Slicing into "Invalid data"
After fixing the model through Netfabb the error is gone and slicing output is provided.

### Project file & How to reproduce

[Ghost_Print_Znap_lab-mik1-back.zip](https://github.com/prusa3d/PrusaSlicer/files/9611056/Ghost_Print_Znap_lab-mik1-back.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Win10

### Printer model

MK3S+MMU2S

stepikovo, 2022-09-28:
Hello @MB3Y, I'm not able to reproduce the issue, it slices just fine, tried in 2.5.0, 2.4.2 and 2.3.3. Can you please elaborate on the issue a bit more? Even a screenshot would be appreciated.

lukasmatena, 2024-02-19:
No response, closing.

closed 2024-02-19 by lukasmatena