prusa3d/PrusaSlicer #8922: Cannot release flatpak version

xarbit, 2022-09-21:
### Description of the bug

After updating all the dependencies used in the department folder, I am unable to release the flatpak version for PrusaSlicer 2.5.0. 

The error is:

```
/usr/lib/gcc/x86_64-unknown-linux-gnu/11.3.0/../../../../x86_64-unknown-linux-gnu/bin/ld: ../libopenvdb.a(points.cc.o): in function `void tbb::detail::d1::small_object_allocator::deallocate<tbb::detail::d1::start_reduce<openvdb::v8_2::tree::LeafManager<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::points::PointDataLeafNode<openvdb::v8_2::PointIndex<unsigned int, 1u>, 3u>, 4u>, 5u> > > const>::LeafRange, openvdb::v8_2::tree::LeafManager<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::points::PointDataLeafNode<openvdb::v8_2::PointIndex<unsigned int, 1u>, 3u>, 4u>, 5u> > > const>::LeafReducer<openvdb::v8_2::tools::count_internal::ActiveVoxelCountOp<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::points::PointDataLeafNode<openvdb::v8_2::PointIndex<unsigned int, 1u>, 3u>, 4u>, 5u> > > > >, tbb::detail::d1::auto_partitioner const> >(tbb::detail::d1::start_reduce<openvdb::v8_2::tree::LeafManager<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::points::PointDataLeafNode<openvdb::v8_2::PointIndex<unsigned int, 1u>, 3u>, 4u>, 5u> > > const>::LeafRange, openvdb::v8_2::tree::LeafManager<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::points::PointDataLeafNode<openvdb::v8_2::PointIndex<unsigned int, 1u>, 3u>, 4u>, 5u> > > const>::LeafReducer<openvdb::v8_2::tools::count_internal::ActiveVoxelCountOp<openvdb::v8_2::tree::Tree<openvdb::v8_2::tree::RootNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::tree::InternalNode<openvdb::v8_2::points::PointDataLeafNode<openvdb::v8_2::PointIndex<unsigned int, 1u>, 3u>, 4u>, 5u> > > > >, tbb::detail::d1::auto_partitioner const>*, tbb::detail::d1::execution_data const&)':
/run/build/PrusaSlicer/deps/build/destdir/usr/local/include/oneapi/tbb/detail/_small_object_pool.h:90: undefined reference to `tbb::detail::r1::deallocate(tbb::detail::d1::small_object_pool&, void*, unsigned long, tbb::detail::d1::execution_data const&)'
collect2: error: ld returned 1 exit status
make[6]: *** [openvdb/openvdb/cmd/CMakeFiles/vdb_print.dir/build.make:104: openvdb/openvdb/cmd/vdb_print] Error 1
make[5]: *** [CMakeFiles/Makefile2:169: openvdb/openvdb/cmd/CMakeFiles/vdb_print.dir/all] Error 2
make[4]: *** [Makefile:146: all] Error 2
make[3]: *** [CMakeFiles/dep_OpenVDB.dir/build.make:86: dep_OpenVDB-prefix/src/dep_OpenVDB-stamp/dep_OpenVDB-build] Error 2
make[2]: *** [CMakeFiles/Makefile2:397: CMakeFiles/dep_OpenVDB.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:404: CMakeFiles/dep_OpenVDB.dir/rule] Error 2
make: *** [Makefile:267: dep_OpenVDB] Error 2
Error: module PrusaSlicer: Child process exited with code 2
```

See : 

- https://github.com/flathub/com.prusa3d.PrusaSlicer/pull/49
- https://buildbot.flathub.org/#/builders/39/builds/13443

### Project file & How to reproduce

regular build on flathub and locally.
 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Linux

### Printer model

N/A

tamasmeszaros, 2022-09-21:
Hi @xarbit !

I am looking into this. In the meantime, could you try to omit dep_TBB from the dependencies build and get TBB from the flatpak infrastructure?

tamasmeszaros, 2022-09-21:
Hi @xarbit !

I am looking into this. You could try to set a different compiler (if there is any other installed on the flathub workers). We are currently using gcc 10.2.1.

xarbit, 2022-09-21:
Ok, any idea what else it can be? 
I tried different Plattform sdk, like 22.08 and 21.08 .. no success

tamasmeszaros, 2022-09-21:
I could make a branch where I update the version of TBB to the newest, let's say `tm_flatpak_tbb_2021.6` and you could try to build that. Otherwise, I have no idea why the link error is happening.

tamasmeszaros, 2022-09-21:
Ok, I've created the branch tm_flatpak_tbb_2021.6 where the build is successful with the newest tbb. You can try if it works. I know it's not an ideal solution as this version was not tested in PrusaSlicer 2.5, but I don't have a better idea.

xarbit, 2022-09-21:
Ok, I'll give that a try tomorrow morning and let you know. Thanks a lot 🙏

xarbit, 2022-09-22:
the build works @tamasmeszaros

tamasmeszaros, 2022-09-22:
I'm glad it does. Please do a bit more aggressive testing, especially for the slicing or CPU intensive operations where this change of TBB could matter.

friedelwolff, 2022-09-22:
I'm cross-posting it here in case it helps:

I tried slicing a few of my previous files, and all crashed so far leaving this on the console:
```
/usr/include/c++/12.1.0/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = Slic3r::SpiralVase; _Dp = std::default_delete<Slic3r::SpiralVase>; typename std::add_lvalue_reference<_Tp>::type = Slic3r::SpiralVase&]: Assertion 'get() != pointer()' failed.
```

xarbit, 2022-09-22:
@tamasmeszaros that was the plan, @friedelwolff  tested and found issues ..

tamasmeszaros, 2022-09-22:
We could do another try: I will try to disable building of vdbprint (which is actually failing in the flathub worker) and then the build might succeed with the currently shipped version of TBB.

xarbit, 2022-09-22:
Ok, just let me know what I need to do. Or will you do it in the same branch and all I need to do is trigger a new build?

tamasmeszaros, 2022-09-23:
So I tried my proposed solution and it did work for me. I've commited it into a separate branch to keep things organized. Branch name is `tm_flatpak_no_vdbprint`. Hope it will work.

xarbit, 2022-09-23:
Great I'll give it a spin later

tamasmeszaros, 2022-09-23:
@friedelwolff Did you test those files on the released AppImages as well? The error message you posted does not suggest any multi-threading issue to me.

xarbit, 2022-09-26:
@tamasmeszaros 
Building from branch tm_flatpak_no_vdbprint failed:

https://buildbot.flathub.org/#/builders/12/builds/13534

xarbit, 2022-09-28:
@tamasmeszaros is there any chance of sharing the code that builds the official AppImage?

tamasmeszaros, 2022-10-03:
Unfortunately, no. But I wrote the part that builds the dependencies and it does the same simple `cmake <PrusaSlicerSrc>/deps` command that is in the flatpak build. It is generally just calling cmake to build the the application, then the rest is assembling the AppImage, does the upload to the internal file servers... etc. Nothing that would help here. 

Our build runs in a docker with Centos7 with 'devtoolset-10' as the toolchain.

tamasmeszaros, 2022-10-03:
Hi @xarbit !

So I did some digging and found out that -flto flags are used in the failing TBB build and the newer version omits -flto flags. 
See this line, which is not present in the newer version. https://github.com/oneapi-src/oneTBB/blob/v2021.5.0/src/tbbmalloc/CMakeLists.txt#L62

This thread tries to deal with the issue https://www.reddit.com/r/Gentoo/comments/6b1dlc/plugin_needed_to_handle_lto_object_and_correct/

You could try if the proposed solution works or not. I could not find any platform where I could reproduce the link error, so I cannot say if this works or not or how to incorporate it into the Flatpak build. 

Good luck!

xarbit, 2022-11-09:
this can be closed

closed 2022-11-09 by lukasmatena