prusa3d/PrusaSlicer #8923: Artillery Sidewinder hangs at the exact same time when printing a file sliced using Prusa Slicer 2.50

azsde, 2022-09-21:
### Description of the bug

When I launch a print of a specific file, the printer always stop at the same time and kill() is called:

[image]

### Project file & How to reproduce

Here's a zip file containing the .stl, the .gcode and and the .3mf

[3D file + gCode.zip](https://github.com/prusa3d/PrusaSlicer/files/9613701/3D.file.%2B.gCode.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Windows 10

### Printer model

Sidewinder X1 - Marlin 2.0.7.2

rtyr, 2022-09-21:
It is most likely caused by incorrect filament settings.

You are printing ASA at 265C with 100% fan speed for the bridges. Your printer likely halts because of thermal runaway protection (unable to reach target temperature).

azsde, 2022-09-21:
Gotcha, I need to look into it, thank you I'll let you know.

rtyr, 2022-09-27:
Can we close the issue?

azsde, 2022-09-27:
Yes, sorry about that your can close it.

closed 2022-09-27 by rtyr