prusa3d/PrusaSlicer #8924: Arachne Engine: add volume flow control like Cura

aimzaimz, 2022-09-21:
**Is your feature request related to a problem? Please describe.**
With the arachne slicing prusaslicer is constantly varying the line width in thin-wall areas. The overall printing speed is kept constant over changing widths resulting in a changing volume flow. Due to the dynamics between extruder and nozzle the flow cannot be changed instantly, hence requested change in line-width/volume flow always comes with a certain delay. This is especially true for bowden-setups or flexible materials. The following screenshot shows that the flowrate is considerably higher in areas with wider lines:
[image]

**Describe the solution you'd like**
Since the volume-flow cannot change with infinite dynamics another approach would be to keep the volume flow constant. This would mean that wider lines have to be printed at a reduced speed and thinner lines have to be printed at high speeds.
A constant volume flow can easily be achieved with any setup (also bowden setups oder flexible materials).
I believe this is also how Cura has the arachne-algorithm implemented. The following screenshot shows the same part and the volume flow is kept constant:
[image]

To achieve that the speed is adjusted:
[image]

**Describe how it would work**
The volume flow is calculated by:
v = Line width * Layer heigth * printing speed

This means for a constant volume flow the printing speed can always be calculated by the formula:
printing speed = volume flow / (line width * layer heigth)

**Describe alternatives you've considered**
-

**Additional context**

Has this approach been considered? At least my setup (bowden) greatly benefits from the cura-implementation.
thanks in advance

CCS86, 2022-11-13:
I see this as a must-have complement to Arachne; even for direct drives.

Merrill added a feature to SS at my request that allows gap-fill to be capped at a percentage of perimeter flow rate. This lets you set a gap fill speed of say 150 mm/s, to let it cruise through the ultra thin areas, and have it reduce speed on a per segment basis when the gap fill gets ultra thick. Very similar end result, but Arachne loses this ability.

stepikovo, 2022-11-23:
Hi guys, have you checked the Pressure equalizer under the _Print settings - Speed_ menu? I think that should solve this issue, let me know what you think.

closed 2026-09-29 by prusaslicer-bot[bot]