prusa3d/PrusaSlicer #8925: Implement some FullControl.xyz features

Fra085, 2022-09-21:
I am sure you have already seen this website before: it enables printing in non-planar and innovative ways, showing us what a 3d printer really can do. The website is FullControl.xyz

While non-planar printing is still challenging for printers because of its mechanical structure, new overhanging techniques are impressive and would be great to implement in PrusaSlicer. In this case, it should be possible to print up to 90 degrees overhangs without any problem, starting from the outer perimeter and offsetting it all around. It would be great in bridging too, maybe building it from the outside-in.

Last, I am also interested in the Ripple Texture Demo they made, apparently giving the perimeter a fuzzy aspect while printing in vase mode. This generates interesting prints, stronger perimeters and fabric-like texture.

Hope to see these features in your slicer or at least to have given you some ideas. If you are already working on this, great and thanks!

kubispe1, 2022-09-21:
Yes, we are watching their challenges. Agree, that there are many interesting ideas.

ElliottBridge, 2022-09-21:
I would love to see this!

closed 2026-09-29 by prusaslicer-bot[bot]